arduino/reference-en #803: No 'keyboard' found.Does the project include '#include <Keyboard.h>'

1399568389, 2020-11-17:
I've loaded the library file for Ridkey.h, but it still gets an error.

The error message:
exit status 1
No 'keyboard' found.Does the project include '#include <Keyboard.h>'?

Short code:
```
#include <Keyboard.h>

void setup() {
// make pin 2 an input and turn on the
// pullup resistor so it goes high unless
// connected to ground:
pinMode(2, INPUT_PULLUP);
Keyboard.begin();
}

void loop() {
//if the button is pressed
if (digitalRead(2) == LOW) {
//Send the message
keyboard.press("Hello!");
}
}
```

per1234, 2020-11-17:
@1399568389 please stop doing this. You're being very disrespectful. If you'll only post on the forum (or even just spend 10 seconds on Google), we'll give you an answer to your problem immediately. But if you persist in this behavior then I'll be forced to ban you and report you to GitHub. It's for you to choose which way you want this to go.

I'll repeat what I said in your last issue (https://github.com/arduino/reference-en/issues/802):

This issue tracker is only to be used to report errors or make feature requests for the [Arduino Language Reference](https://www.arduino.cc/reference/en/). It is **NOT** a place to request assistance with your project. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules. Be sure to post your code, using code tags (`</>` button on the toolbar).

closed 2020-11-17 by per1234

1399568389, 2020-11-17:
@per1234 The problem is that all my other file calls are fine except this one，I also submitted questions to arduino community, and others recommended me to ask them here.

1399568389, 2020-11-17:
I'm very sorry to trouble you.at the first time, I misunderstood that the format of my code was not added, so I added the content again and uploaded it again，Sorry again

per1234, 2020-11-17:
>I also submitted questions to arduino community, and others recommended me to ask them here.

Please post a link to where they recommended this so I can educate them on this matter.