arduino/reference-en #819: Clarify function of `serialEvent` on Arduino Due

maltekliemann, 2021-03-01:
According to the [docs for serialEvent](https://forum.arduino.cc/index.php?topic=389856.0):

> `serialEvent()`, `serialEvent1()``serialEvent2()`, [sic] and `serialEvent3()`  don't work on the Arduino Due.

This sentence needs clarification. What does "don't work" mean? Are they not implemented on the Arduino Due? Are they no-op? Does using them cause undefined behavior?

To give this question some context: I'm using `serialEvent` on a Arduino Due, and I'm only experiencing expected behavior. Now I'm worried I may be getting UB.

per1234, 2021-03-01:
Hi @maltekliemann. Thanks for reporting this error. This information is incorrect. Arduino Due has support for all those. I submitted a pull request to correct this error:
https://github.com/arduino/reference-en/pull/800
If you want to take a look at my proposed changes, I welcome any feedback.

maltekliemann, 2021-03-01:
Hey @per1234, thank you very much! The PR obviously clarifies my issue.

closed 2021-03-01 by maltekliemann