arduino/reference-en #827: Incorrect breadcrumb links

monodop, 2021-05-03:
I am noticing that the breadcrumb links don't seem to be working correctly for me.

For example, on the page: https://www.arduino.cc/reference/en/language/functions/communication/serial/readbytes/

The link to "Serial" in the breadcrumbs points to: https://www.arduino.cc/reference/en#serial

I believe, instead, it should point to: https://www.arduino.cc/reference/en/language/functions/communication/serial/

I've noticed this happen in the past as well, but I don't have all of the examples on hand right now.

[image]

VeloSteve, 2024-04-10:
Still a problem in 2024.  For example, if I am on https://www.arduino.cc/reference/en/language/functions/communication/stream/streamreadbytesuntil/
and I click Stream in the breadcrumb line it takes me to
https://www.arduino.cc/reference/en/
This is just a list of some Arduino functions, and not the Stream information that a user would expect.

Examining this page, there is a Stream link which goes to the correct page (https://www.arduino.cc/reference/en/language/functions/communication/stream/)  but requiring this extra step is a bug.

By the way, it's only a coincidence that I was on a Stream function like the original poster.  This also happens with Wire, Mouse, SPI, and probably all the others.

MaderDash, 2024-06-30:
> Still a problem in 2024. For example, if I am on https://www.arduino.cc/reference/en/language/functions/communication/stream/streamreadbytesuntil/ and I click Stream in the breadcrumb line it takes [...]

Just a bump on this, IDK if it's just me or not. But it is working properly for me. Please let me know if this is still an issue.

MaderDash, 2024-06-30:
> I am noticing that the breadcrumb links don't seem to be working correctly for me.
>
> For example, on the page: https://www.arduino.cc/reference/en/language/functions/communication/serial/readbytes/ [...]

Same statement. This may be fixed, please check to see if this is still an issue please?

per1234, 2024-06-30:
@MaderDash the behavior is exactly as described in the bug report for me. When I click the "Serial" link in the breadcrumbs, I am taken to the URL https://www.arduino.cc/reference/en#serial, which is simply the Arduino Language Reference home page (not even scrolled down to the relevant section, since there is no `serial` anchor in the page).

monodop, 2024-06-30:
@MaderDash I still see the same issue with the breadcrumbs today

monodop, 2024-06-30:
[image]

VeloSteve, 2024-09-12:
Yes, the same issue from 2024 is still there.  The link takes you to the index instead of the topic.