arduino/reference-en #830: switch...case reference confusing

sterretjeToo, 2021-05-22:
https://www.arduino.cc/reference/en/language/structure/control-structure/switchcase/

Allowed data types according to the reference are _int_ and _char_. However, any integral type is allowed (e.g. uint32_t, byte, ...).

MalcolmBoura, 2021-05-23:
A few points:
1. Should mention including the stdint.h header file. Really Arduino.h should do that, see next point.
2. In my opinion, int, unsigned etc should be deprecated because what you actually get depends on the platform. The types from stdint.h should be the default.
3. The case values must be constants.
4. When falling through from one case to the next then a comment, eg "// fall through", should usually be provided to indicate that the break has been omitted deliberately.

sterretjeToo, 2021-05-27:
@MalcolmBoura,

Regarding (1, "mention including stdint.h"), you will have to keep the intended audience in mind; it's beginners in the language and they don't really care. I would prefer the format to be that of Linux man pages ;) 

Regarding (4), there is no fall-through example so the point is a little moot. It's also the user's responsibility to document those cases ;)

karlsoderby, 2023-11-20:
Hey guys, and sorry for the **late** response 😅 you are indeed accurate that any integral type is supported, and I have updated the article for that (here: https://github.com/arduino/reference-en/pull/937).

closed 2023-11-20 by karlsoderby