arduino/reference-en #836: tone(pin, frequency, duration) seems to ignore 'duration' (Nano clone)

brewmanz, 2021-06-09:
On my Nano clone, calling 
`tone(pin, frequency, duration)`
[https://www.arduino.cc/reference/en/language/functions/advanced-io/tone/](url)
seems to ignore duration. e.g. changing the [https://www.arduino.cc/en/Tutorial/BuiltInExamples/toneMelody](url) 'loop' to
```
void loop() {
  // slide notes; plays as expected
  for (float thisNote = 64.0; thisNote < 4096.0; thisNote *= 1.01) { // 64 - 4096 = 6 octaves, *1.01 = ~70 per octave, ~ 420 notes
    tone(pinTune, (int)thisNote);
    delay(10);
  }
  noTone(pinTune);
  delay(500);
  // slide notes; plays way too fast 
  for (float thisNote = 64.0; thisNote < 4096.0; thisNote *= 1.01) { // 64 - 4096 = 6 octaves, *1.01 = ~70 per octave, ~ 420 notes
    tone(pinTune, (int)thisNote, 10);
  }
  noTone(pinTune);
  delay(1000);
}
```
Should produce identical glissandos, but produces a 'slow' one (as expected, and then a 'fast' one, which seems to indicate that 'duration' is being ignored (no delay).

Compile & Upload log ..
```
/home/bryan/git/Arduino/build/linux/work/arduino-builder -dump-prefs -logger=machine -hardware /home/bryan/git/Arduino/build/linux/work/hardware -hardware /home/bryan/.arduino15/packages -tools /home/bryan/git/Arduino/build/linux/work/tools-builder -tools /home/bryan/git/Arduino/build/linux/work/hardware/tools/avr -tools /home/bryan/.arduino15/packages -built-in-libraries /home/bryan/git/Arduino/build/linux/work/libraries -libraries /home/bryan/git/BrewArduino/libraries -fqbn=arduino:avr:nano:cpu=atmega328 -vid-pid=1A86_7523 -ide-version=10814 -build-path /tmp/arduino_build_696342 -warnings=default -build-cache /tmp/arduino_cache_2015 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/home/bryan/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/home/bryan/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avr-gcc.path=/home/bryan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=/home/bryan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.arduinoOTA.path=/home/bryan/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/home/bryan/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -verbose /home/bryan/git/BrewArduino/ArduinoSensorsFolder/BuzzersEtcFolder/toneMelody/toneMelody.ino
/home/bryan/git/Arduino/build/linux/work/arduino-builder -compile -logger=machine -hardware /home/bryan/git/Arduino/build/linux/work/hardware -hardware /home/bryan/.arduino15/packages -tools /home/bryan/git/Arduino/build/linux/work/tools-builder -tools /home/bryan/git/Arduino/build/linux/work/hardware/tools/avr -tools /home/bryan/.arduino15/packages -built-in-libraries /home/bryan/git/Arduino/build/linux/work/libraries -libraries /home/bryan/git/BrewArduino/libraries -fqbn=arduino:avr:nano:cpu=atmega328 -vid-pid=1A86_7523 -ide-version=10814 -build-path /tmp/arduino_build_696342 -warnings=default -build-cache /tmp/arduino_cache_2015 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/home/bryan/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/home/bryan/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avr-gcc.path=/home/bryan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=/home/bryan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.arduinoOTA.path=/home/bryan/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/home/bryan/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -verbose /home/bryan/git/BrewArduino/ArduinoSensorsFolder/BuzzersEtcFolder/toneMelody/toneMelody.ino
Using board 'nano' from platform in folder: /home/bryan/.arduino15/packages/arduino/hardware/avr/1.8.3
Using core 'arduino' from platform in folder: /home/bryan/.arduino15/packages/arduino/hardware/avr/1.8.3
Detecting libraries used...
/home/bryan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10814 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/bryan/.arduino15/packages/arduino/hardware/avr/1.8.3/cores/arduino -I/home/bryan/.arduino15/packages/arduino/hardware/avr/1.8.3/variants/eightanaloginputs /tmp/arduino_build_696342/sketch/toneMelody.ino.cpp -o /dev/null
Generating function prototypes...
/home/bryan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10814 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/bryan/.arduino15/packages/arduino/hardware/avr/1.8.3/cores/arduino -I/home/bryan/.arduino15/packages/arduino/hardware/avr/1.8.3/variants/eightanaloginputs /tmp/arduino_build_696342/sketch/toneMelody.ino.cpp -o /tmp/arduino_build_696342/preproc/ctags_target_for_gcc_minus_e.cpp
/home/bryan/git/Arduino/build/linux/work/tools-builder/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /tmp/arduino_build_696342/preproc/ctags_target_for_gcc_minus_e.cpp
Compiling sketch...
/home/bryan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10814 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/bryan/.arduino15/packages/arduino/hardware/avr/1.8.3/cores/arduino -I/home/bryan/.arduino15/packages/arduino/hardware/avr/1.8.3/variants/eightanaloginputs /tmp/arduino_build_696342/sketch/toneMelody.ino.cpp -o /tmp/arduino_build_696342/sketch/toneMelody.ino.cpp.o
Compiling libraries...
Compiling core...
Using precompiled core: /tmp/arduino_cache_2015/core/core_arduino_avr_nano_cpu_atmega328_b379999e9df8609848927a438c5693f3.a
Linking everything together...
/home/bryan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-gcc -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega328p -o /tmp/arduino_build_696342/toneMelody.ino.elf /tmp/arduino_build_696342/sketch/toneMelody.ino.cpp.o /tmp/arduino_build_696342/../arduino_cache_2015/core/core_arduino_avr_nano_cpu_atmega328_b379999e9df8609848927a438c5693f3.a -L/tmp/arduino_build_696342 -lm
/home/bryan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-objcopy -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0 /tmp/arduino_build_696342/toneMelody.ino.elf /tmp/arduino_build_696342/toneMelody.ino.eep
/home/bryan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-objcopy -O ihex -R .eeprom /tmp/arduino_build_696342/toneMelody.ino.elf /tmp/arduino_build_696342/toneMelody.ino.hex
/home/bryan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-size -A /tmp/arduino_build_696342/toneMelody.ino.elf
Sketch uses 3624 bytes (11%) of program storage space. Maximum is 30720 bytes.
Global variables use 60 bytes (2%) of dynamic memory, leaving 1988 bytes for local variables. Maximum is 2048 bytes.
/home/bryan/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude -C/home/bryan/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf -v -patmega328p -carduino -P/dev/ttyUSB0 -b115200 -D -Uflash:w:/tmp/arduino_build_696342/toneMelody.ino.hex:i 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/bryan/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf"
         User configuration file is "/home/bryan/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyUSB0
         Using Programmer              : arduino
[... cut ...]
avrdude: 3624 bytes of flash written
avrdude: verifying flash memory against /tmp/arduino_build_696342/toneMelody.ino.hex:
avrdude: load data flash data from input file /tmp/arduino_build_696342/toneMelody.ino.hex:
avrdude: input file /tmp/arduino_build_696342/toneMelody.ino.hex contains 3624 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.47s

avrdude: verifying ...
avrdude: 3624 bytes of flash verified

avrdude done.  Thank you.
```

per1234, 2021-06-09:
Hi @brewmanz. Thanks for taking the time to submit an issue.

The `tone()` function is not ignoring the `duration` parameter, it's just not using it as you expect. This function is non-blocking. That means that after you call `tone()`, the sketch execution continues, so you can do other things while the tone is playing. If you have set a duration, the tone output will stop after that duration has passed. So in your sketch the tone frequency is incrementing as fast as the `for` loop can iterate.

I see this non-blocking duration behavior is not mentioned in[ the `tone()` page](https://www.arduino.cc/reference/en/language/functions/advanced-io/tone/) of the Arduino Language Reference. We can use this issue to track that deficiency.

kengdahl, 2021-06-28:
@per1234 I'll bring this to the attention of the content team so that we can add an explanation in the documentation.