arduino/reference-en #849: Proposal: `BTN_BUILTIN`

zfields, 2021-09-14:
There is a precedent for LEDs which have been built into the board and are configurable by the user to be named `LED_BUILTIN`. This is a de facto standard and has been accepted broadly across the community.

New boards are coming out with button built into the board, which can be configured by the user. There are conflicting names currently, such as ESP's `BUTTON_BUILTIN` or STM's `USER_BTN`. I believe `BUTTON_BUILTIN` follows the naming precedent better than `USER_BTN`. I would like to see Arduino take the lead here and dictate a standard name for such a feature, and bring alignment to the Arduino ecosystem.

`BTN_BUILTIN` seems like a good compromise. `BTN` is a reasonable abbreviation for button, and it would make `BTN_BUILTIN` the same number of characters as `LED_BUILTIN`. The name also expresses that the button is a feature of the board and not simply and external addition.

It would much easier to port Arduino sketches across hardware platforms if Arduino would take a stance and add `BTN_BUILTIN` to the standard or specification.

`BUTTON_BUILTIN` References:
- https://github.com/espressif/arduino-esp32/blob/master/variants/micro_s2/pins_arduino.h#L75
- https://github.com/esp8266/Arduino/blob/master/variants/espectro/pins_arduino.h#L40
- https://github.com/f32c/arduino/blob/master/hardware/fpga/f32c/variants/FleaFPGA-Uno/variant.h#L17

`USER_BTN` References:
- https://github.com/stm32duino/Arduino_Core_STM32/pull/1490#issuecomment-915874601
- https://github.com/search?q=org%3Astm32duino+USER_BTN&type=code

_If this is not the appropriate place for such a suggestion, please point me in the right direction._

karlsoderby, 2023-11-16:
Hi @zfields , just stumbled upon this so my apologies for the 2+ years wait. I believe this would be a great implementation, but the buttons on all (official) Arduino boards are used for resetting purposes only, where a single tap resets, double tap places it in bootloader mode and so on.

As essentially no official Arduino boards would benefit from it, I think adding it to the reference might be the wrong approach (and as of now, we don't have the actual API implemented). I think it is something "board specific", so a good place would be to create an example and submit it to the official core repository for the supported board, or create an individual library for it (which is maybe a bit overkill 😉 ).

I'm closing this issue, but thanks again for a great suggestion! Feel free to reopen it again.

closed 2023-11-16 by karlsoderby

zfields, 2023-11-16:
Thank you for responding. Late is better than never, and I do appreciate it.

I understand you have no need of the feature at this time, but I think it would be good for Arduino to offer a framework, or guidance, for those who wish extend the base API.

In this examle, the hope would be that both ESP and STM would have arrived at the same (or similar) name by following clear guidance from an Arduino specification document.