arduino/reference-en #918: ArduinoBLE: mention that you need to discoverService() before querying characteristic

nevyn, 2023-07-13:
At: https://www.arduino.cc/reference/en/libraries/arduinoble/bledevice.characteristic/

It would be helpful to mention in the main documentation body for `BLEDevice.characteristic()` that the API requires you to either `BLEDevice.discoverAttributes()` or `BLEDevice.discoverSerivce()` first. It is hinted at from the example, but it's really easy to miss.

It could also be mentioned in the conceptual documentation over at `https://www.arduino.cc/reference/en/libraries/arduinoble/` that characteristics need to be discovered before being queried. There are sections about notifying and writing characteristics, but no section about basic reading of characteristics; that could be a good place to mention it.

Thanks!

per1234, 2023-07-13:
Thanks for your suggestion @nevyn. This repository is only used for development of the content of the [**Arduino Language Reference**](https://www.arduino.cc/reference/en/). Since this issue is about the ArduinoBLE library's documentation, it is off topic for this issue tracker.

Library reference content is hosted in each individual library repository. For the ArduinoBLE library, it is here:

https://github.com/arduino-libraries/ArduinoBLE/tree/master/docs

Please submit this as an issue or pull request to the `arduino-libraries/ArduinoBLE` repository:

https://github.com/arduino-libraries/ArduinoBLE

closed 2023-07-13 by per1234

nevyn, 2023-07-14:
@per1234 thanks! The docs site does link to this repo for reporting documentation issues. It says: "Find anything that can be improved? Suggest corrections and new documentation via GitHub." With a link here. I recommend changing that link to the correct repo when reading library documentation off of arduino.cc!

nevyn, 2023-07-14:
Re-reported as https://github.com/arduino-libraries/ArduinoBLE/issues/306