arduino/reference-en #924: bitRead and bitWrite documentation ambiguity

mjbmikeb2, 2023-09-01:
bitRead says "Reads a bit of a number."
Is that all signed and unsigned numbers of any size?

bitWrite says, "Writes a bit of a numeric variable"
Again, is that all signed and unsigned numbers of any size?

According to this thread
https://forum.arduino.cc/t/bitread-and-bitwrite-for-numbers-bigger-than-a-byte/1160471
bitWrite doesn't work for 64 bit numbers.

I don't know if this is a bug or a feature. Either way can the documentation please be updated to say what datatypes are currently supported.

Thanks.

karlsoderby, 2023-11-17:
Hi @mjbmikeb2 , thanks for raising the issue. I looked into it, and made some changes to the documentation that hopefully explains things a bit better:
- Max size to read (64 bits)
- Max size to write (32 bits)
- Added an example of how to actually read each individual bit, and print out the raw binary (hopefully the example itself adds some value on how the `bitRead()` function works as well.
- `float` / `double` are not supported.

Changes here: https://github.com/arduino/reference-en/pull/933

mjbmikeb2, 2023-11-19:
Typo in https://www.arduino.cc/reference/en/language/functions/bits-and-bytes/bitwrite/

"variables up to an unsigned long (32 bits / 8 bytes)."

Should say 4 bytes.

"variables up to an unsigned long (32 bits / 4 bytes)."

karlsoderby, 2024-01-08:
Hi there @mjbmikeb2 , thanks for spotting it, just been corrected! https://github.com/arduino/reference-en/pull/960